russhughes/gc9a01_mpy #6: UnicodeError

mytechnotalent, 2022-09-28:
I am getting a `UnicodeError` when trying to open this as a bytes file.  I am trying to convert these to `.py` files.  Here is the related issue which routes me back here as a bug.

[BUG](https://github.com/peterhinch/micropython_data_to_py/issues/1 )

```
with open("dc540_logo.jpg", "rb") as f:
    b = f.read()  # Image as an array of bytes
    display.jpg(b, 0, 0)
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 109, in <module>
UnicodeError: 
>>> 
```

russhughes, 2022-09-28:
The jpg method expects the filename of the jpg to draw.

Use: 

`display.jpg("dc540_logo.jpg", 0, 0)` or `display.jpg("dc540_logo.jpg", 0, 0, gc9a01.SLOW)` if you do not have enough free memory to buffer the entire jpg.

mytechnotalent, 2022-09-28:
The issue I am having is converting it to a conversion script to convert the jpg to `.py` file and use your `display.jpg` to call it.  

https://github.com/peterhinch/micropython_data_to_py/issues/1

I am unable to do this.

russhughes, 2022-09-29:
You could use utils/imgtobitmap.py to create a python module from the jpg and use the bitmap method to display it.

mytechnotalent, 2022-09-29:
Thanks @russhughes!  Would the `imgtobitmap.py` literally make a `image.py` from from the `.jpg`?

russhughes, 2022-10-02:
Yes, look at the updated examples/rp2/toasters for an example.

mytechnotalent, 2022-10-03:
@russhughes the issue that I had with `imgtobitmap.py` was that the image was of poor color depth.  The `bits_per_pixel` can go up to 8 (I believe) and it as I said does not appear well.

Is there a way to get a better quality image from `imgtobitmap.py`?

`imgtobitmap image_file bits_per_pixel >image.py`

mytechnotalent, 2022-10-03:
@russhughes 
I am attempting to use the `imgtobitmap.py` and I was able to get a memoryview object and a successful `.py` file however when I import the file and then try to use it.

```
>>> import dc540_logo
>>> display.bitmap(dc540_logo, 0, 0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert NoneType to int
```

mytechnotalent, 2022-10-03:
truncated contents of `dc540_logo.py`

```HEIGHT = 240
WIDTH = 240
COLORS = 256
BITS = 460800
BPP = 8
PALETTE = [0x0000,0xa631,0x5cef,0xaa52,0x4942,0x4429,0xa210,0xbad6,0x9ef7,0x4c6b,0x1bef,0xd9e6,0xae73,0x58ce,0xb6b5,0x0742,0xb294,0x37ce,0xe641,0x2b6b,0x54ad,0x0321,0x508c,0x33ad,0x083a,0x2f8c,0xd5c5,0xd1a4,0xe839,0xca62,0xcd83,0x073a,0x0c5b,0x0319,0x0b5b,0x1bdf,0xec5a,0xe320,0x58ee,0x54cd,0x1cdf,0xb2c4,0x97de,0x0f7c,0x17be,0xaeab,0x139d,0x50b4,0x33cd,0x2fac,0xb6e5,0x93bd,0xe418,0x37e6,0xb3c5,0x4cb3,0x0419,0xcaaa,0x51c4,0xfcde,0xe841,0x8fa4,0x107c,0x8f9c,0xbddf,0x349d,0xfcee,0x18be,0x875a,0x2bab,0xf07b,0x8b83,0xf8bd,0xe420,0x2fc4,0x8b7b,0xca9a,0xf49c,0x8339,0xcec3,0xf8c5,0x2a8b,0x8762,0x8341,0xc6b1,0xec62,0xb7f6,0x34e5,0x4c9b,0xf0b3,0x32e5,0x6962,0xb3e5,0xf4cc,0xd083,0x69b2,0x27aa,0xf4a4,0xf3e4,0x299a,0xdbbe,0xf8ed,0x6662,0x8bb3,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000,0x0000]
_bitmap =\
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
...
BITMAP = memoryview(_bitmap)
```

mytechnotalent, 2022-10-10:
@russhughes almost there!  It is coming up now black and white instead of color but working!

```
from machine import Pin, SPI 

from gc9a01 import GC9A01
import dc540_logo

spi = SPI(0, baudrate=40000000, polarity=0, sck=Pin(6), mosi=Pin(7))
display = GC9A01(spi, 240, 240, backlight=Pin(2, Pin.OUT), reset=Pin(14, Pin.OUT), dc=Pin(15, Pin.OUT))
display.init()

display.bitmap(dc540_logo, 0, 0)
```

mytechnotalent, 2022-10-10:
`python imagetobitmap.py dc540_logo.jpg 5 >dc540_logo.py`

This is the command I am using to make a .py.

mytechnotalent, 2022-10-10:
Also getting memory allocation error.
```>>> import main
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 10, in <module>
MemoryError: memory allocation failed, allocating 115200 bytes
>>> 
```

russhughes, 2022-10-18:
Can you post the dc540_logo.jpg file?

mytechnotalent, 2022-10-18:
@russhughes https://ibb.co/6Y2Zhb6

russhughes, 2022-10-19:
The bitmap requires more memory than was available.  See the new `pbitmap` method that can draw a bitmap line by line that is larger than available memory. I was able to draw your converted logo with 8 bits per pixel.  To have enough memory, you will need to precompile the converted bitmap module using mpy-cross.

[image]

closed 2022-10-19 by russhughes

mytechnotalent, 2022-10-19:
Thanks @russhughes!

mytechnotalent, 2022-10-21:
@russhughes 
```
>>> import main
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 12, in <module>
KeyError: MAP
>>>```

```
  1 from machine import Pin, SPI
  2 # from utime import sleep
  3 from gc9a01 import GC9A01
  4
  5 import dc540_logo
  6
  7 # display config
  8 spi = SPI(0, baudrate=40000000, polarity=0, sck=Pin(6), mosi=Pin(7))
  9 display = GC9A01(spi, 240, 240, backlight=Pin(2, Pin.OUT), reset=Pin(14, Pin.OUT), dc=Pin(15, Pin.OUT))
 10 display.init()
 11
 12 display.pbitmap(dc540_logo, 0, 0, 0)```

russhughes, 2022-10-21:
Sorry, the new firmware files were in different folders. Reflash with a firmware.bin from the latest update, and it should work.

mytechnotalent, 2022-10-21:
Works great @russhughes!  We are looking to use this driver on our local Defcon Chapter (not the official Defcon) for a badge.  Last year we did a Tarot Badge and a Tree Of Life Badge before that.  Your work is the best as I thank you greatly for your driver in C for MP.

russhughes, 2022-10-21:
Thanks, glad you are finding it useful.