russhughes/gc9a01_mpy #7: Problem compiling.

bhrobinson, 2022-10-10:
I am sure I am doing something wrong, but I am trying to compile the for an RP2040 board, and getting very strange results. 

I was using you 7789 instructions, and tried to use for this was well. 

I get the following issues::

/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:121:6: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'print'
  121 |     .print = mp_file_print,
      |      ^~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:121:14: error: initialization of 'unsigned char' from 'void (*)(const mp_print_t *, void *, mp_print_kind_t)' {aka 'void (*)(const struct _mp_print_t *, void *, mp_print_kind_t)'} makes integer from pointer without a cast [-Werror=int-conversion]
  121 |     .print = mp_file_print,
      |              ^~~~~~~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:121:14: note: (near initialization for 'mp_file_type.slot_index_make_new')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:121:14: error: initializer element is not computable at load time
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:121:14: note: (near initialization for 'mp_file_type.slot_index_make_new')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:122:6: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'locals_dict'
  122 |     .locals_dict = (mp_obj_dict_t *)&mp_file_locals_dict,
      |      ^~~~~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:122:20: error: initialization of 'unsigned char' from 'mp_obj_dict_t *' {aka 'struct _mp_obj_dict_t *'} makes integer from pointer without a cast [-Werror=int-conversion]
  122 |     .locals_dict = (mp_obj_dict_t *)&mp_file_locals_dict,
      |                    ^
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:122:20: note: (near initialization for 'mp_file_type.slot_index_print')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:122:20: error: initializer element is not computable at load time
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c:122:20: note: (near initialization for 'mp_file_type.slot_index_print')
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/firmware.dir/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/mpfile.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c: In function 'write_spi':
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:59:67: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'protocol'
   59 |     mp_machine_spi_p_t *spi_p = (mp_machine_spi_p_t*)spi_obj->type->protocol;
      |                                                                   ^~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c: At top level:
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1483:6: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'print'
 1483 |     .print = gc9a01_GC9A01_print,
      |      ^~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1483:14: error: initialization of 'unsigned char' from 'void (*)(const mp_print_t *, void *, mp_print_kind_t)' {aka 'void (*)(const struct _mp_print_t *, void *, mp_print_kind_t)'} makes integer from pointer without a cast [-Werror=int-conversion]
 1483 |     .print = gc9a01_GC9A01_print,
      |              ^~~~~~~~~~~~~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1483:14: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_make_new')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1483:14: error: initializer element is not computable at load time
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1483:14: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_make_new')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1484:6: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'make_new'
 1484 |     .make_new = gc9a01_GC9A01_make_new,
      |      ^~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1484:17: error: initialization of 'unsigned char' from 'void * (*)(const mp_obj_type_t *, size_t,  size_t,  void * const*)' {aka 'void * (*)(const struct _mp_obj_type_t *, unsigned int,  unsigned int,  void * const*)'} makes integer from pointer without a cast [-Werror=int-conversion]
 1484 |     .make_new = gc9a01_GC9A01_make_new,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1484:17: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_print')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1484:17: error: initializer element is not computable at load time
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1484:17: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_print')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1485:6: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'locals_dict'
 1485 |     .locals_dict = (mp_obj_dict_t*)&gc9a01_GC9A01_locals_dict,
      |      ^~~~~~~~~~~
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1485:20: error: initialization of 'unsigned char' from 'mp_obj_dict_t *' {aka 'struct _mp_obj_dict_t *'} makes integer from pointer without a cast [-Werror=int-conversion]
 1485 |     .locals_dict = (mp_obj_dict_t*)&gc9a01_GC9A01_locals_dict,
      |                    ^
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1485:20: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_call')
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1485:20: error: initializer element is not computable at load time
/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c:1485:20: note: (near initialization for 'gc9a01_GC9A01_type.slot_index_call')
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/firmware.dir/Users/bhrobinson/git/Pico/usermodules/gc9a01_mpy/src/gc9a01.c.obj] Error 1
make[2]: *** [CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [all] Error 2
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1

russhughes, 2022-10-10:
New changes in MicroPython are causing the issue. I'll try and get this fixed today.

mytechnotalent, 2022-10-10:
Thank you both.  I am experiencing it as well.  Thank you @russhughes you have one of the finest libraries I have ever seen.  I am still having issues converting images to `.py` format but I will continue to try to figure it out.

russhughes, 2022-10-10:
This should be fixed, it compiles now. I don't have my display with me so I'm not able to test it.

closed 2022-10-10 by russhughes

mytechnotalent, 2022-10-10:
I will try it now.

mytechnotalent, 2022-10-10:
Works @russhughes 
```
from machine import Pin, SPI 

from gc9a01 import GC9A01
import dc540_logo

spi = SPI(0, baudrate=40000000, polarity=0, sck=Pin(6), mosi=Pin(7))
display = GC9A01(spi, 240, 240, backlight=Pin(2, Pin.OUT), reset=Pin(14, Pin.OUT), dc=Pin(15, Pin.OUT))
display.init()

display.bitmap(dc540_logo, 0, 0, 0)
```

bhrobinson, 2022-10-10:
That is awesome, man! I spent all weekend as I am just learning this part, and figured it was me

Thank you so much!

commit 61d2c142 (2022-10-10): fix #7 due to optimised mp_obj_type_t representations change