russhughes/gc9a01_mpy #11: Unable to display two screens at once

mytechnotalent, 2023-02-15:
`
import _thread
from machine import Pin, SPI
from utime import sleep
from gc9a01 import GC9A01, YELLOW

import NotoSansMono_32 as font
import dc540_logo

semaphore_lock = _thread.allocate_lock()

# display config
spi0 = SPI(0, baudrate=4000000, polarity=0, sck=Pin(6), mosi=Pin(7))
display0 = GC9A01(spi0, 240, 240, backlight=Pin(2, Pin.OUT), reset=Pin(14, Pin.OUT), dc=Pin(15, Pin.OUT))
display0.init()

spi1 = SPI(1, baudrate=4000000, polarity=0, sck=Pin(10), mosi=Pin(11))
display1 = GC9A01(spi1, 240, 240, backlight=Pin(3, Pin.OUT), reset=Pin(0, Pin.OUT), dc=Pin(1, Pin.OUT))
display1.init()

# button config
left_button = Pin(19, Pin.IN, Pin.PULL_UP)
right_button = Pin(18, Pin.IN, Pin.PULL_UP)
up_button = Pin(17, Pin.IN, Pin.PULL_UP)
down_button = Pin(16, Pin.IN, Pin.PULL_UP)

# clear screen
display0.fill_rect(0, 0, 240, 240, 0)
display1.fill_rect(0, 0, 240, 240, 0)

# display logo
display0.pbitmap(dc540_logo, 0, 0, 0)
display1.pbitmap(dc540_logo, 0, 0, 0)

# sleep
sleep(2)

# clear screen
display0.fill_rect(0, 0, 240, 240, 0)
display1.fill_rect(0, 0, 240, 240, 0)

# display text
display0.write(font, 'main menu', 35, 60, YELLOW)
display1.write(font, 'hello', 10, 100, YELLOW)

class Button:
        def __init__(self):
                self.left_pressed_state = False
                self.right_pressed_state = False
                self.core1_in_use = False

        def left_pressed(self, pin):
                print('state: ' + str(self.core1_in_use))
                if self.core1_in_use:
                        left_button.irq(handler=None)
                        semaphore_lock.acquire()
                        if left_button.value() and not self.left_pressed_state:
                                self.left_pressed_state = True
                                print('breaky')
                                _thread.exit()
                                self.core1_in_use = False
                        elif not left_button.value() and self.left_pressed_state:
                                self.left_pressed_state = False
                        left_button.irq(handler=self.left_pressed)
                        semaphore_lock.release()

        def right_pressed(self, pin):
                print('state: ' + str(self.core1_in_use))
                if not self.core1_in_use:
                        right_button.irq(handler=None)
#                       semaphore_lock.acquire()
                        if right_button.value() and not self.right_pressed_state:
                                self.right_pressed_state = True
                                Flag.clear_run_flag()
                                print('start thread')
                                self.core1_in_use = True
                                _thread.start_new_thread(demo.play, ())
                        elif not right_button.value() and self.right_pressed_state:
                                self.right_pressed_state = False
                        right_button.irq(handler=self.right_pressed)
#                       semaphore_lock.release()

def up_button_pressed(pin):
        print('up')

class Demo:
        def play(self):
                while True:
                        semaphore_lock.acquire()
                        print('demo')
                        sleep(1)
                        print('1st')
                        sleep(1)
                        print('2nd')
                        sleep(1)
                        print('3rd')
                        sleep(1)
                        print('4th')
                        sleep(1)
                        semaphore_lock.release()

button = Button()
demo = Demo()

left_button.irq(trigger=Pin.IRQ_FALLING, handler=button.left_pressed)
right_button.irq(trigger=Pin.IRQ_FALLING, handler=button.right_pressed)
up_button.irq(trigger=Pin.IRQ_FALLING, handler=up_button_pressed)

while True:
        print('main')
        sleep(1)
`

The left screen illuminates and the right screen does not.  I switched the wires and the screens however spi1 does not work.

mytechnotalent, 2023-02-15:
https://cdn.discordapp.com/attachments/1012726491756371989/1075231027330560070/IMG_2763.jpg
https://cdn.discordapp.com/attachments/1012726491756371989/1075231040555192390/IMG_2764.jpg
https://cdn.discordapp.com/attachments/1012726491756371989/1075231555045298207/3656C9FE-DBB9-46C1-9DA5-E77A97E15573.jpg

russhughes, 2023-02-15:
I have not tried it, but driving two displays at the same time should work. I can look into it this weekend.

mytechnotalent, 2023-02-15:
Thank you @russhughes!  Above is the exact code I am using.

mytechnotalent, 2023-02-17:
@russhughes  in addition to the two displays I am trying to write a scroll function as I saw your example from bottom to top but I wanted a scroll from right to left however my example is choppy and not smooth

`   def scroll(self, delay=0.001):
        self.display.fill_rect(0, 0, 240, 240, 0)
        text = "Hello, world long string please help as it is does not work."
        text_width = len(text)

        # Set the initial x-coordinate to the right edge of the display
        x = text_width

        while x >= -text_width:
            # Clear the previous position of the text
            self.display.fill_rect(x, 120, text_width, font.HEIGHT, gc9a01.BLACK)
            # self.display.fill_rect(0, 0, 240, 240, 0)

            # Draw the text at the current position
            self.display.write(font, text, x, 120, gc9a01.WHITE)

            # Move the text one pixel to the left
            x -= 1

            # Wait for the specified delay
            utime.sleep(delay)`

russhughes, 2023-02-17:
My scroll demo uses the hardware scrolling feature that only works on that axis.

mytechnotalent, 2023-02-17:
no worries @russhughes I will work with that access then.  Thanks again for looking into the dual screens this weekend.

russhughes, 2023-02-18:
It works, see the hello_world.py demo program: https://github.com/russhughes/gc9a01_mpy/blob/893b8584a5ef752dc810e2a8d65f2bf3f9ecbffc/examples/RP2/hello_world.py

[image]

mytechnotalent, 2023-02-18:
@russhughes is it perhaps my config?
```spi0 = SPI(0, baudrate=4000000, polarity=0, sck=Pin(6), mosi=Pin(7))
display0 = GC9A01(spi0, 240, 240, backlight=Pin(2, Pin.OUT), reset=Pin(14, Pin.OUT), dc=Pin(15, Pin.OUT))
display0.init()

spi1 = SPI(1, baudrate=4000000, polarity=0, sck=Pin(10), mosi=Pin(11))
display1 = GC9A01(spi1, 240, 240, backlight=Pin(3, Pin.OUT), reset=Pin(0, Pin.OUT), dc=Pin(1, Pin.OUT))
display1.init()```

mytechnotalent, 2023-02-18:
Got it working @russhughes thank you!!!

closed 2023-02-18 by mytechnotalent

russhughes, 2023-02-19:
Excellent!

mytechnotalent, 2023-02-21:
> My scroll demo uses the hardware scrolling feature that only works on that axis.

@russhughes would you be able to implement a scroll from right to left rather than top to bottom?  Such that it would print a larger sentence scrolling from right to left?

russhughes, 2023-02-22:
Not easily, there is no full-screen framebuffer, and the hardware only supports scrolling on a single axis. Can you rotate the display 90' to get right and left scrolling?

mytechnotalent, 2023-02-22:
possibly