russhughes/gc9a01_mpy #12: Pure C version for Pico

mytechnotalent, 2023-03-11:
HI @russhughes currently working on another project this one using the C sdk without MicroPython.  Do you have a version of the gc9a01 to work in C as the one you have here is technically C but with all of the MicroPython functions to port to a MicroPython environment.

Thanks as always!

russhughes, 2023-03-11:
I don't have any drivers that work outside of MicroPython.

mytechnotalent, 2023-03-11:
No worries thanks anyway.  I am trying to develop a simple one to display a single pixel but does not work 

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/spi.h"

// Pin Definitions
#define PIN_LCD_DC 6
#define PIN_LCD_CS 5
#define PIN_LCD_RST 7
#define PIN_LCD_BL 8
#define PIN_LCD_MOSI 3
#define PIN_LCD_SCLK 2

// LCD commands
#define LCD_CMD_SOFT_RST 0x01
#define LCD_CMD_SLEEP_OUT 0x11
#define LCD_CMD_COL_ADDR_SET 0x2A
#define LCD_CMD_PAGE_ADDR_SET 0x2B
#define LCD_CMD_MEM_WR 0x2C
#define LCD_CMD_PIXEL_FMT_SET 0x3A
#define LCD_CMD_DISP_ON 0x29
#define LCD_CMD_MADCTL 0x36

// LCD parameters
#define LCD_WIDTH 240
#define LCD_HEIGHT 240
#define LCD_PIXEL_SIZE 3

// Buffer for storing pixel data
uint8_t lcd_buffer[LCD_WIDTH * LCD_HEIGHT * LCD_PIXEL_SIZE];

// Look-up table for 16-bit to 18-bit color mapping
const uint8_t color_map[32] = {
    0x00, 0x08, 0x10, 0x18, 0x20, 0x29, 0x31, 0x39, 0x41, 0x49, 0x52, 0x5A, 0x62, 0x6A, 0x73, 0x7B,
    0x83, 0x8B, 0x94, 0x9C, 0xA4, 0xAC, 0xB4, 0xBC, 0xC5, 0xCD, 0xD5, 0xDD, 0xE5, 0xEE, 0xF6, 0xFE
};

// Function to write a command to the LCD
void lcd_write_cmd(uint8_t cmd)
{
    gpio_put(PIN_LCD_DC, 0); // set DC pin low to indicate command mode
    spi_write_blocking(spi0, &cmd, 1); // send command over SPI
}

// Function to write data to the LCD
void lcd_write_data(uint8_t *data, int len)
{
    gpio_put(PIN_LCD_DC, 1); // set DC pin high to indicate data mode
    spi_write_blocking(spi0, data, len); // send data over SPI
}

// Function to set a pixel to a specific color
void set_pixel(uint16_t x, uint16_t y, uint32_t color)
{
    uint8_t r = color >> 11; // extract red value from 16-bit color
    uint8_t g = (color >> 5) & 0x3F; // extract green value from 16-bit color
    uint8_t b = color & 0x1F; // extract blue value from 16-bit color

    // Expand color to 18-bit using look-up table
    r = color_map[r];
    g = color_map[g];
    b = color_map[b];

    uint32_t index = (y * LCD_WIDTH + x) * LCD_PIXEL_SIZE; // calculate index in
    // calculate index in buffer for given pixel
    lcd_buffer[index] = r;
    lcd_buffer[index + 1] = g;
    lcd_buffer[index + 2] = b;
}

// Function to initialize the LCD
void lcd_init()
{
    // Initialize pins
    gpio_init(PIN_LCD_DC);
    gpio_set_dir(PIN_LCD_DC, GPIO_OUT);
    gpio_init(PIN_LCD_CS);
    gpio_set_dir(PIN_LCD_CS, GPIO_OUT);
    gpio_init(PIN_LCD_RST);
    gpio_set_dir(PIN_LCD_RST, GPIO_OUT);
    gpio_init(PIN_LCD_BL);
    gpio_set_dir(PIN_LCD_BL, GPIO_OUT);
    gpio_init(PIN_LCD_MOSI);
    gpio_set_function(PIN_LCD_MOSI, GPIO_FUNC_SPI);
    gpio_init(PIN_LCD_SCLK);
    gpio_set_function(PIN_LCD_SCLK, GPIO_FUNC_SPI);

    // Set default pin states
    gpio_put(PIN_LCD_DC, 0);
    gpio_put(PIN_LCD_CS, 1);
    gpio_put(PIN_LCD_RST, 1);
    gpio_put(PIN_LCD_BL, 1);

    // Initialize SPI
    spi_init(spi0, 8000000);
    spi_set_format(spi0, 8, SPI_CPOL_0, SPI_CPHA_0, SPI_MSB_FIRST);
    spi_set_slave(spi0, false);

    // Perform soft reset
    gpio_put(PIN_LCD_RST, 0);
    sleep_ms(10);
    gpio_put(PIN_LCD_RST, 1);
    sleep_ms(10);

    // Initialize LCD
    lcd_write_cmd(LCD_CMD_SLEEP_OUT); // turn off sleep mode
    lcd_write_cmd(LCD_CMD_COL_ADDR_SET); // set column address range
    lcd_write_data((uint8_t[]){0x00, 0x00, 0x00, 0xEF}, 4);
    lcd_write_cmd(LCD_CMD_PAGE_ADDR_SET); // set page address range
    lcd_write_data((uint8_t[]){0x00, 0x00, 0x01, 0x3F}, 4);
    lcd_write_cmd(LCD_CMD_PIXEL_FMT_SET); // set pixel format to 18 bits per pixel
    lcd_write_data((uint8_t[]){0x66}, 1);
    lcd_write_cmd(LCD_CMD_MADCTL); // set display orientation
    lcd_write_data((uint8_t[]){0x00}, 1);
    lcd_write_cmd(LCD_CMD_DISP_ON); // turn on display
}

// Function to update the LCD with current buffer
void lcd_update()
{
    lcd_write_cmd(LCD_CMD_MEM_WR); // enable memory write mode
    lcd_write_data(lcd_buffer, LCD_WIDTH * LCD_HEIGHT * LCD_PIXEL_SIZE); // send pixel data
}

int main()
{
    stdio_init_all(); // initialize stdio
    lcd_init(); // initialize LCD
    // Draw a gradient
    for (int y = 0; y < LCD_HEIGHT; y++) {
        for (int x = 0; x < LCD_WIDTH; x++) {
            uint32_t color = ((x & 0xF8) << 8) | ((y & 0xFC) << 3) | ((x & 0x07) << 3) | ((y & 0x03) << 1);
            set_pixel(x, y, color);
        }
    }

    lcd_update(); // update LCD with buffer

    while (1) {
        // main loop
    }
    return 0;
}
```

closed 2023-03-11 by mytechnotalent