russhughes/gc9a01_mpy #13: Can this repo wokrs on esp32c3?

jd3096-mpy, 2023-04-27:
I try to build firmware ,build success but start failed

russhughes, 2023-04-27:
It should work.  I have compiled firmware for the GENERIC_C3 and GENERIC_C3_USB boards but have not tested them.  I

jd3096-mpy, 2023-04-28:
> It should work. I have compiled firmware for the GENERIC_C3 and GENERIC_C3_USB boards but have not tested them. I

I found out the reason.
I use latest micropython 1.20.0 repo to build this firmware,but it seems they changed the way to add extra c modules.

Any third-party C extensions will need to be updated to work with this change. See commits https://github.com/micropython/micropython/commit/662b9761b37b054f08fe2f7c00d0fce3a418d0b0 and https://github.com/micropython/micropython/commit/3ac8b5851e5f4dade465d52b91ed2ccc17851263 for details.

[https://github.com/micropython/micropython/releases](url)

jd3096-mpy, 2023-04-28:
> It should work. I have compiled firmware for the GENERIC_C3 and GENERIC_C3_USB boards but have not tested them. I

Thanks for your firmware,it test it and it works good.
[image]

But I still need time to checkout why the firmware build by myself started faild,you can close this issue.

closed 2023-04-28 by russhughes