russhughes/gc9a01_mpy #16: Display only working when code is executed from Thonny or run manually from REPL via serial.

ciobania, 2023-05-26:
Hello, 

sorry to bother you with another "issue".
As the title says, I've successfully integrated with your library, and all seemed to work just fine, and was able to progress with my project.

I wanted it to run from autostart, via `main.py`. I reboot the board, and surprise and shock ensued. The backlight is on, the display flickers for a split second or so (probably because of `tft.init()` and `tft.fill()) than nothing else is displayed.

Do you have any experience with this kind of issue, or behaviour? Do I need to add something extra, or add more sleeps in?

I read some topics on the internet, where people have added a `time.sleep(1.5)` before initializing the OLED displays. I tried that too, in hope that it'll allow me to stop pulling my hair out.

I'm not sure what else to try out.

thank you kindly for any information or suggestion you might have!

russhughes, 2023-05-26:
I have not run into that issue. By reboot, do you mean a soft reset (ctrl-d), hardware reset, or power cycle? 

I usually have a main.py that contains a short sleep followed by an `import application`. The short sleep gives me time to ctl-c and stop the program if something is wrong with the application.

ciobania, 2023-05-26:
Soft reset, or power cycle.

I'll try to put it back again in it's own module and import it via `main.py`
I'm just puzzled by it. I have the exact same display as the one in photo and Pi PicoW.

I'll try to get creative with the sleeps, and run one of your examples to see if I get a different behaviour.

I'll keep this issue on for now, maybe I can provide some feedback, and close it if I don't get progress.

Thank you for getting back at me.

ciobania, 2023-05-26:
Just thought to add my code, maybe I'm missing something, or doing something wrong.
I used SPI0 for this, maybe that's the issue...

```
'''
Draw a dashboard for aquarium stats using the slower but less memory intensive method
of blitting each Minimum Coded Unit (MCU) block. Usually 8×8 pixels but can
be other multiples of 8.

GC9A01 display connected to a Raspberry Pi Pico.

Pico Pin   Display
=========  =======
14 (GP10)  BL  - 9  (GP6)
15 (GP11)  RST - 10 (GP7)
16 (GP12)  DC  - 6  (GP4)
17 (GP13)  CS  - 7  (GP5)
18 (GND)   GND - 8  (GND)
19 (GP14)  CLK - 4  (GP2)
20 (GP15)  DIN - 5  (GP3)
'''

import gc
import random
import time
import utime
from machine import Pin, SPI
import gc9a01
# import vga1_bold_16x32 as font
import vga1_8x8 as font8
import vga1_8x16 as font
# import chango_16 as font_16
# import chango_32 as font_32
# import chango_64 as font_64
time.sleep(5)
gc.enable()
gc.collect()

def cycle(p):
    '''
    return the next item in a list
    '''
    try:
        len(p)
    except TypeError:
        cache = []
        for i in p:
            yield i
            cache.append(i)
        p = cache
    while p:
        yield from p

time.sleep(5)
spi = SPI(0, baudrate=60000000, sck=Pin(2), mosi=Pin(3))
tft = gc9a01.GC9A01(
    spi,
    240,
    240,
    reset=Pin(7, Pin.OUT),
    cs=Pin(5, Pin.OUT),
    dc=Pin(4, Pin.OUT),
    backlight=Pin(6, Pin.OUT),
    rotation=0)

# enable display and clear screen
time.sleep(5)
tft.init()
tft.fill(gc9a01.BLACK)
width = tft.width()

# cycle thru jpg's
col_max = tft.width() - font.WIDTH*6
row_max = tft.height() - font.HEIGHT

read_temperature = 20
sunrise = '03:59'
sunset = '19:57'

time.sleep(2)
while True:
    time.sleep(.5)
    now = time.localtime()
    now_time = "{:02d}:{:02d}:{:02d}".format(now[3], now[4], now[5])
    now_date = "{:02d}.{:02d}.{:04d}".format(now[2], now[1], now[0])
    try:
        tft.text(font, now_time, 85, 8) # 12 chars
        tft.text(font, now_date, 75, 25) # 12 chars
        tft.hline(35, 45, col_max - 30, gc9a01.WHITE)
        tft.vline(100, 45, row_max - 30, gc9a01.WHITE)
        tft.text(font, "{:.02f} °C".format(read_temperature), 30, 60) # 12 chars
        tft.jpg('sunrise_1.jpg', 110, 50, gc9a01.SLOW)
        tft.text(font, sunrise, 110, 115) # 12 chars
        tft.jpg('sunsets.jpg', 110, 140, gc9a01.SLOW)
        tft.text(font, sunset, 110, 205) # 12 chars

        utime.sleep(0.25)
    except Exception as _:
        with open('error.txt', 'a') as stream:
            stream.write(f'{now_time} - Error was: {_}\n')

```

russhughes, 2023-05-26:
The power cycle issue is strange.  I could see soft resets leaving hardware peripherals like the SPI in an odd state.  Something like this:

```
try:
  spi = SPI(0, baudrate=60000000, sck=Pin(2), mosi=Pin(3))
  tft = gc9a01.GC9A01(
    spi,
    240,
    240,
    reset=Pin(7, Pin.OUT),
    cs=Pin(5, Pin.OUT),
    dc=Pin(4, Pin.OUT),
    backlight=Pin(6, Pin.OUT),
    rotation=0)
  
    ... more code ...

finally:
  spi.deinit()
```

Will ensure that the spi is deinit'ed at the end of the program or before a soft reset.
I haven't used the library with a PicoW only a Pico, I'll try it this weekend.

ciobania, 2023-05-26:
Ok, I've tried one of your examples, `scroll.py`, and I get the same behavior.
The display is showing data, the text is scrolling if I run it from Thonny, and NO output when rebooted via a power cycle and attempting to run via `main.py`.
It also works just fine if I drop in `shell repl` via Thonny, and type: 
```
import main
```

I do see the screen flicker once.

ciobania, 2023-05-28:
Hi, 

Did you manage to get anywhere with it? Do you get the same behavior?

Thank you.

russhughes, 2023-05-28:
I'm not seeing any issues with my Pico or Pico W using the original or the new MicroPython 1.20 firmware I uploaded yesterday.  I am using the same pin wiring as my demo code shows.

ciobania, 2023-05-28:
Is there a reason why it wouldn't work with the wiring I chose, basically it's still SPI, just the 0 not the 1.

Thanks for uploading v1.20 firmware. Will try that later, to see if there's any change. I cannot change the pins, to use the ones you're using, without redoing, resoldering the whole board. Is the driver `gc9a01` using hardcoded pins, does it work only with those pins that you selected?

russhughes, 2023-05-28:
No, the driver is not hard coded to those pins and can use other pins.

ciobania, 2023-05-28:
Ok. 

Thanks for trying, and releasing the library with MPy v1.20.

I've tried on a separate PicoW, and I have the same behavior. Not sure why that is...

I ordered another type of display and will seek other libraries to see if I have better luck.
Much appreciated

closed 2023-05-28 by ciobania

ciobania, 2023-05-28:
Scratch that, it works now...

The issue was bad wiring, between pins RES and BLK! (facepalm) (facepalm)

russhughes, 2023-05-28:
Excellent!