russhughes/gc9a01_mpy #17: Can't compile firmware for ESP8266

richyguitar, 2023-06-07:
Hello there,
first of all thanks a lot for the provided code, I really appreciate your effort!

Since I am a noob when it comes to compiling in general and I am merely able to write a simple script in Python, I'd like to reach out to you. I am using a Wemos D1 mini ESP8266 board.

I got the compiling the firmware running with the docker container using the following input and I was able to add the cexample code from micropython and successfully tested the cexample class:
`docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk make -j 4 V=1 USER_C_MODULES=/home/richy/micropython/examples/usercmodule/`

Now I tried adding your gc9a01 library using the following input:
`docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk make V=1 USER_C_MODULES=/home/richy/micropython/drivers/gc9a01_mpy-main`

I get the following error message (see the full output in the file attached):

> xtensa-lx106-elf-ld: build-GENERIC/firmware.elf section '.text' will not fit in region 'iram1_0_seg'
> xtensa-lx106-elf-ld: region 'iram1_0_seg' overflowed by 15976 bytes
> build-GENERIC/src/gc9a01.o:(.text.gc9a01_GC9A01_draw+0x1c): undefined reference to 'lround'
> build-GENERIC/src/gc9a01.o: In function 'gc9a01_GC9A01_draw':
> /home/richy/micropython/drivers/gc9a01_mpy-main/src/gc9a01.c:474: undefined reference to 'lround'
> /home/richy/micropython/drivers/gc9a01_mpy-main/src/gc9a01.c:476: undefined reference to 'lround'
> /home/richy/micropython/drivers/gc9a01_mpy-main/src/gc9a01.c:478: undefined reference to 'lround'
> /home/richy/micropython/drivers/gc9a01_mpy-main/src/gc9a01.c:489: undefined reference to 'lround'
> make: *** [Makefile:231: build-GENERIC/firmware.elf] Error 1

Please advise.

Thanks a lot in advance,
Richard

[compile_output.txt](https://github.com/russhughes/gc9a01_mpy/files/11677544/compile_output.txt)

richyguitar, 2023-06-23:
Hello there,
still very much appreciate your work.

If anyone is interested and stumbles across this issue searching for a similar thing:
In the meantime I ran across LVGL which includes a GC9A01-driver and a powerful library which seems to fit my needs. I was able to compile the firmware using lv_micropython (an example implementation of LVLGL bindings for micropython using version 4 of the ESP-IDF and release 7 of lv_micropython running `git clone -b v4.0 --recursive https://github.com/espressif/esp-idf/` and `git clone -b release/v7 https://github.com/lvgl/lv_micropython` (see https://github.com/lvgl/lv_micropython/blob/master/ports/esp32/README.md#setting-up-the-toolchain-and-esp-idf and https://github.com/lvgl/lv_micropython (be aware to select the right version of lv_micropython for the right building instructions)).

Cheers

closed 2023-06-23 by richyguitar

liopoos, 2025-02-24:
> Hello there, still very much appreciate your work.
>
> If anyone is interested and stumbles across this issue searching for a similar thing: In the meantime I ran across LVGL which includes a [...]

Hello! According to your method, I successfully compiled the firmware of lv micropython, and I also saw that the lv binding contains a driver for gc9a01, but I don't know how to use it. I want to implement correct screen lighting, can I refer to your code? Thank you very much!