russhughes/gc9a01_mpy #24: rp2040 waveshare with 1.28 round display not recognice firmware

jajawunderbar, 2023-10-27:
Hi, i want to use it on the above rp2040 module, but the rp2040 dont boot after copy the firmware on the drive. With other normal rp2040 firmware its no problem. So there is a display coming with it, but its a normal rp2040. Thanks Mathias :)

ovjanman, 2023-11-06:
Is there any chance you could build and upload the updated firmware for the rp2040? I'm very new to all of this and really struggling to build it myself on windows

russhughes, 2023-11-08:
@ovjanman Updated firmware just uploaded.

jajawunderbar, 2023-11-08:
Thanks!, but my rp2040 again doesnt restart. With other firmware it does?!

russhughes, 2023-11-08:
I tired the firmware.uf2 with a T-PicoC3 and it copied, rebooted and gave me a MicroPython REPL. I did find this [discussion](https://github.com/orgs/micropython/discussions/11174#discussioncomment-5497129) so I recompiled with PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64. Give the newest firware.uf2 a try and let  me know if it works any better.

jajawunderbar, 2023-11-08:
Thanks! Unfortunately, it still doesn't work. It ignores the firmware.

russhughes, 2023-11-09:
Is the working MicroPython firmware from the MicroPython website?

jajawunderbar, 2023-11-09:
Yes. i tried it just again. I chose the firmware for the normal rp2040 on the mp site and it runs.
https://micropython.org/download/RPI_PICO/

russhughes, 2023-11-09:
Excellent.

ovjanman, 2023-11-09:
Thankyou very much @russhughes. Will give it a go myself later this week. Appreciate all the effort