russhughes/gc9a01_mpy #26: Accessing the fonts of the pre-build Firmwares

Fjordx, 2023-11-09:
The readme states: The firmware directory contains pre-compiled firmware ... with ...frozen python font files.

Now, how do you access them?
I tried -  **import  NotoSans_32 as font0** but get an import error : ImportError: no module named 'NotoSans_32'
I'm using the RP2 Firmware.

Btw: the example images in the readme of the vector fonts are broken

russhughes, 2023-11-09:
I've uploaded the missing images. The fonts take a lot of flash so only some of the fonts are frozen into the firmware. You can see which fonts are frozen by typing `help('modules')` in the REPL.

Fjordx, 2023-11-12:
Thanks a lot for the fast reply! I'll leave this issue open for future devs to stumble upon till the readme includes this info or till i generated a pull request. Hope that#s alright!