russhughes/gc9a01_mpy #30: draw_arc is missing

JonHylands, 2024-02-27:
Flashed new firmware this morning, fill_arc works fine, draw_arc appears to be missing.

```
>>> dir(GC9A01)
['__class__', '__name__', 'write', '__bases__', '__dict__', 'arc', 'bitmap', 'blit_buffer', 'bounding', 'circle', 'draw', 'draw_len', 'fill', 'fill_arc', 'fill_circle', 'fill_polygon', 'fill_rect', 'hard_reset', 'height', 'hline', 'init', 'inversion_mode', 'jpg', 'line', 'map_bitarray_to_rgb565', 'off', 'offset', 'on', 'pbitmap', 'pixel', 'polygon', 'polygon_center', 'rect', 'rotation', 'set_window', 'sleep_mode', 'soft_reset', 'text', 'vline', 'vscrdef', 'vscsad', 'width', 'write_len']
```

JonHylands, 2024-02-27:
I have to say, Russ, this is brilliant work - I've been building up a full touch-based GUI system based on your driver. I just switched from the Seeed round display to the Waveshare ESP32-S3 version, installed your brand new build, and I'm very happy.

[image]
[image]

JonHylands, 2024-02-27:
I see the issue - the MicroPython function is arc, not draw_arc as the documentation says.

russhughes, 2024-02-27:
I’ll get the docs fixed. arc_widths > 1 didn’t make the cut as I got very busy at work and the project has been in limbo for a while. I’ve been getting requests for a newer firmware version so I released what I had ready. Definitely let me know if you run into other inconsistencies.

JonHylands, 2024-02-27:
Thanks Russ, I appreciate that. I'll let you know if I run into anything else.

commit 24799e4d (2024-02-27): Fix #30 arc misnamed draw_arc
closed 2024-02-27 by russhughes