russhughes/gc9a01_mpy #35: tft_config.py for M5Dial?

amplatzer, 2024-03-29:
Hi,
I tried to create the tft_config.py for M5Dial, but failed. 

according to this link .https://github.com/Gitshaoxiang/M5Dial-Gui-Demo/blob/40357725d0c812131e28543a6e4f23a7981d6282/components/esp_lvgl_port/examples/touchscreen/main/main.c#L37

/* LCD settings */
#define EXAMPLE_LCD_SPI_NUM         (SPI3_HOST)
#define EXAMPLE_LCD_PIXEL_CLK_HZ    (40 * 1000 * 1000)
#define EXAMPLE_LCD_CMD_BITS        (8)
#define EXAMPLE_LCD_PARAM_BITS      (8)
#define EXAMPLE_LCD_COLOR_SPACE     (ESP_LCD_COLOR_SPACE_BGR)
#define EXAMPLE_LCD_BITS_PER_PIXEL  (16)
#define EXAMPLE_LCD_DRAW_BUFF_DOUBLE (1)
#define EXAMPLE_LCD_DRAW_BUFF_HEIGHT (50)
#define EXAMPLE_LCD_BL_ON_LEVEL     (1)

/* LCD pins */
#define EXAMPLE_LCD_GPIO_SCLK       (GPIO_NUM_7)
#define EXAMPLE_LCD_GPIO_MOSI       (GPIO_NUM_6)
#define EXAMPLE_LCD_GPIO_RST        (GPIO_NUM_48)
#define EXAMPLE_LCD_GPIO_DC         (GPIO_NUM_4)
#define EXAMPLE_LCD_GPIO_CS         (GPIO_NUM_5)
#define EXAMPLE_LCD_GPIO_BL         (GPIO_NUM_45)

I created this , but not working, can someone helps, TIA. 

`
    spi = SPI(1, baudrate=40000000, sck=Pin(7), mosi=Pin(6))
    return gc9a01.GC9A01(
        spi,
        240,
        240,
        reset=Pin(48, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(4, Pin.OUT),
        backlight=Pin(45, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size
    )

`

russhughes, 2024-04-02:
I don't have one to test with, but looking at the [schematic on the M5Stack website](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/M5Dial/Sch_M5Dial.pdf)
I would suggest trying:

```
spi = SPI(1, baudrate=40000000, sck=Pin(6), mosi=Pin(5))
return gc9a01.GC9A01(
  spi,
  240,
  240,
  reset=Pin(8, Pin.OUT),
  cs=Pin(7, Pin.OUT),
  dc=Pin(4, Pin.OUT),
  backlight=Pin(9, Pin.OUT),
  rotation=rotation,
  options=options,
  buffer_size=buffer_size
)
```

amplatzer, 2024-04-04:
> I don't have one to test with, but looking at the [schematic on the M5Stack website](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/M5Dial/Sch_M5Dial.pdf) I would suggest [...]

But that's not real GPIO # , is it?   I thought it's defined by M5Stack  G1 ~ G15

amplatzer, 2024-04-12:
> I don't have one to test with, but looking at the [schematic on the M5Stack website](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/M5Dial/Sch_M5Dial.pdf) I would suggest [...]

thanks,, this works.

russhughes, 2024-04-12:
Excellent.

closed 2024-04-23 by russhughes