russhughes/gc9a01_mpy #38: Please may I ask a question?

peterhinch, 2024-05-10:
I am writing a MicroPython driver for this display. The aim is compatibility with my GUI's e.g. [nano-gui](https://github.com/peterhinch/micropython-nano-gui). My initialisation is based on yours. It works, but I would like to know _how_ it works. Your initialisation writes to a number of registers that don't appear in the datasheet. I have commented out the code for most of these registers with no ill effect, but I am left with [registers 66h and 67h](https://github.com/russhughes/gc9a01_mpy/blob/3d6441ea4e596905254ba889c43b6e40c9af32f4/src/gc9a01.c#L2614) which are necessary.

Where can I find documentation on these registers - and other missing registers? Is there a source for the data values written?

russhughes, 2024-05-10:
I would also like to know this. I have yet to find any information on the undocumented registers; some don't appear in the Datasheets, and others don't have any details. The only information I've seen suggests a display manufacturer provided the init parameters.

peterhinch, 2024-05-11:
Thanks for letting me know. I'm familiar with the _values_ of args being supplied by display manufacturers but undocumented registers are new to me. Ugh.

I'm rather bemused by the fact that I could comment out most of these assignments. However there are two caveats to that observation. Firstly I've only run brief functionality tests: more comprehensive tests may reveal shortcomings. Secondly, to conserve RAM my driver uses 4-bit colors, expanded to 16 bits via a LUT. Consequently a screen can only display 16 colors. Color accuracy is therefore not an issue. This may explain why some of those "secret" settings have no observable effect.

closed 2024-05-11 by peterhinch