russhughes/gc9a01_mpy #42: Trouble compiling for a RP2040

martyphee, 2024-07-07:
The docs say to use this command

```
cd micropython/ports/rp2
make \
    BOARD=RPI_PICO \
    FROZEN_MANIFEST=../../../../gc9a01c/manifest.py \
    USER_C_MODULES=../../../gc9a01c/src/micropython.cmake \
    submodules clean all
```

I think that directory is wrong if you just use a git clone it would be `gc9a01_mpy`

I got past that but get this error now

```
Root pointer registrations updated
[  5%] Generating genhdr/root_pointers.h
freeze error executing "../../../gc9a01_mpy/manifest.py": Error in manifest file: /Users/marty/develop/micro/micropython/gc9a01_mpy/manifest.py: [Errno 2] No such file or directory: '/Users/marty/develop/micro/micropython/gc9a01_mpy/manifest.py'
make[3]: *** [CMakeFiles/BUILD_FROZEN_CONTENT] Error 1
make[2]: *** [CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[1]: *** [all] Error 2
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1
```
Full output: [output.log](https://github.com/user-attachments/files/16118666/output.log)

Using this command from `./micropython/ports/rp2`

```
make \
    BOARD=RPI_PICO \
    PICO_TOOLCHAIN_PATH=/Applications/ARM/bin \
    FROZEN_MANIFEST=../../../gc9a01_mpy/manifest.py \
    USER_C_MODULES=../../../gc9a01_mpy/src/micropython.cmake \
    submodules clean all
```

Computer: MBP M3

joshwand, 2024-08-18:
I had this same problem, and I fixed it by using absolute paths instead of relative paths.

Also, for apple silicon, when building `mpy-cross`, (which you *must* do before doing the firmware build), you may need to prefix the build command: `arch -arch x86_64 make -C mpy-cross clean all`

martyphee, 2024-08-24:
Thank you. That worked for me also.

closed 2024-08-24 by martyphee