russhughes/gc9a01_mpy #47: jpg() fast mode ask for  too much memory

amplatzer, 2024-10-01:
115200 for 240*240 is way too much for micropython. 

And the slow mode is too slow. 

The M5 libraries load the jpg file blazing fast and with less memory.   how did they do that?

russhughes, 2024-10-10:
Can you post a link to the M5 Libraries you are looking at?

amplatzer, 2024-10-11:
> Can you post a link to the M5 Libraries you are looking at?

https://github.com/m5stack/M5GFX/tree/master
https://github.com/m5stack/M5Dial

but it's in C, they offer the pre-compiled MPY firmware with M5burner, so I can't find the source code in python.

amplatzer, 2024-10-11:
I was testing with M5dial which has a ESP32-S3FN8 chip, and without PSRAM, the offical firmware displays the jpg files fast.

amplatzer, 2024-10-11:
> Can you post a link to the M5 Libraries you are looking at?

ah, looks like this is the one you need , https://github.com/m5stack/uiflow-micropython