russhughes/gc9a01_mpy #52: RP2350 support

zacharesmer, 2025-02-27:
In case anyone else is searching, I've compiled this for RP2350 (pico 2) and it seems to be working!

RP2350 support was added to micropython in [1.24](https://github.com/micropython/micropython/releases/tag/v1.24.0) so it requires that or newer.

The command to build it is exactly the same as for RP2040 but with BOARD=RPI_PICO2. This is what I ran from the `micropython/ports/rp2` directory:

```
make BOARD=RPI_PICO2 FROZEN_MANIFEST=/my/path/to/gc9a01_mpy/manifest.py USER_C_MODULES=/my/path/to/gc9a01_mpy/src/micropython.cmake 
```

It's slightly different from the documentation because on my computer the RP2040 command in the docs stopped with the error `Internal build error: The submodule list should not be empty`. I ended up running `git submodule update --init` in the micropython repo first, then omitting the `submodules` argument and using the command above. This is what [this part of the micropython docs](https://github.com/micropython/micropython/wiki/Getting-Started) says to do. But the docs also say to do it the other way [here](https://docs.micropython.org/en/latest/develop/gettingstarted.html). There's some discussion [here](https://forums.raspberrypi.com/viewtopic.php?t=383517) as well. I'm not sure if this is a change in micropython or an issue with my environment, but maybe it'll be useful for someone else having a similar problem to see this. 

If you want, I can make a PR to add compiled firmware and add the RP2350 instructions to the docs, but I'm guessing you might want to make your own binaries so you know what's in there, and I'm not sure quite how the docs are built. 

Thanks!

ErikOostveen, 2025-03-13:
I would love to have the gc9a01 firwmare for the Pico 2 ... can you make it available?

ilbuonmarcio, 2025-03-22:
Yes, please, it would be awesome to have the firmware already compiled if possible

PR #53 (2025-03-24): Add RPI_PICO2 and RPI_PICO2_W binaries
Add uf2 files for Pico 2 and Pico 2 W, built with the most recent version of micropython pulled from Github (commit [f1018ee](https://github.com/micropython/micropython/commit/f1018ee5c2dd0eabcc0cfe7e02c607594e489788))

Fixes #52, partially. This does not add instructions to the docs or binaries for other RP2350 boards.
closed 2025-03-24 by russhughes